rusefi/alphax-2chan #11: rev.A schematics review

andreika-git, 2021-12-25:
@ElDominio Can you please review the schematics?

[alphax_2ch.pdf](https://github.com/rusefi/alphax-2chan/files/7776154/alphax_2ch.pdf)

ElDominio, 2021-12-25:
Looks fine to me!

mck1117, 2021-12-26:
We should go ahead and at least connect `IN_CRANK-` to `U2` pin 3, and `U2` pin 4 to `IN_D3`.  This will get us at least the same impedance on both VR pins - important that it gets clipped symmetrically, or it can push the VR signal "off center".

If the area is available, we should add a second pullup so we get a nearly-free extra digital input channel as well.

andreika-git, 2021-12-27:
Updated the schematics:
- more stm32_stop friendly;
- controlling the analog power supply (TLS115) and bluetooth module supply;
- misc. minor improvements;
[alphax_2ch.pdf](https://github.com/rusefi/alphax-2chan/files/7781608/alphax_2ch.pdf)

mck1117, 2021-12-27:
Dunno if we care, but we could merge Q2 and Q15 and have only a single pullup control for both VR/hall switchable inputs, since there's no way to use only one as hall at a time.

ElDominio, 2021-12-27:
Some late 90s/early 2000s cars are VR crank/Hall cam, would they still work?

rusefillc, 2021-12-28:
@ElDominio yes they would. the discussion is about VR+ and VR- both becoming separate hall inputs if you have hall crank.

andreika-git, 2021-12-28:
@ElDominio @rusefillc The board is ready for review!
[image]

Qwerty-OFF, 2021-12-30:
Немного мелочей по печатной плате:
1. У U1 включить сплошную заливку полигона к центральному паду
2. Обратить немного внимания на DRC, там есть ругательства на шелкографию, которые возможно легко пофиксятся
3. Инвертировать цвета QR кода на плате, иначе может плохо распознаваться (только если это не белая плата с черной шелкографией). Также стоит сделать его из картинки более высокого разрешения, чтобы не было видно артефактов конвертации (сглаживания углов). Как пример вот:
[image]
4. Добавить переходных отверстий GND, чтобы не было бесполезной висящей в воздухе меди в ситуациях подобных этой:
[image]
И стараться избегать появления таких отростков там где они не нужны
[image]
[image]
5. Попутно где глаз зацепится можно исправить такие некрасивости
[image]
6. У транзисторов для чуть более лучшего теплоотвода можно сделать полигон к переходным отверстиям, или разместить их непосредственно в паде (только небольшого диаметра, чтобы припой не утекал)
[image]

andreika-git, 2021-12-31:
Update:
- better QR codes and other minor improvements (thanks @Qwerty-OFF)
- move VR_ANALOG from SENS1 to RES3 (both ADC3)
P.S. The board files are ready for BOM review!

closed 2022-01-12 by rusefillc