rusefi/alphax-2chan #12: Usb power no DFU no reset

rusefillc, 2022-01-12:
[image]

andreika-git, 2022-01-12:
Current status: L101 should be populated, otherwise STM32 doesn't start.
If L101 is populated, then TLS115 should NOT be populated

andreika-git, 2022-01-12:
https://community.st.com/s/question/0D50X00009XkhxN/stm32f407vgt6-adc-pins-unused-vrefvrefvddavssa

mck1117, 2022-01-12:
That's a design mistake - the TLS115 shouldn't power 5VA.  It should power things like the onboard pullups, but not any onboard sensors or logic. For example, right now if you short out the 5v sensor supply, it will incapacitate the STM32's oscillator!

mck1117, 2022-01-12:
I guess on this revision we will have to run without the TLS and with inductor installed.

andreika-git, 2022-01-13:
One of the ways to use TLS115 on the rev.A is to cut the +5VA trace going to the input module, and have both L101 and TLS populated.
[image]
In that case:
- The external sensors will be powered from TLS115 and will be protected;
- The STM32 analog core and the internal MAP sensor will be powered from the internal +5VA via L101;
- The input module pullups will be powered internally (which affects only 2Step input);
- Other pullups (CLT, IAT, CAM) will be powered from TLS115.

Also we have another option of cutting a different trace which gives us all pullups and the internal MAP sensor powered from TLS115:
[image]

rusefillc, 2022-01-13:
Plan is to go with 504-PTSA080515V012 fuse for these five boards

closed 2022-02-16 by rusefillc