rusefi/alphax-2chan #14: BT instructions

rusefillc, 2022-01-15:
We have BT on-board

Any initialization/setup instructions? Anything we need to change in https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/hellen/alphax-2chan?

rusefillc, 2022-01-17:
```
After the WL9981TC Bluetooth transparent transmission module is powered on and initialized, it outputs "READY!" through the serial port, indicating that the module is working normally, and automatically
Automatically send broadcast messages, and the module is in a connectable state at this time.

WL9981TC Bluetooth BLE transparent transmission module
READY! \r\n (initialization complete, connection allowed)
4.4 Establishing a connection
When the WL9981TC is in normal working state, it automatically sends broadcast messages, and the APP of the mobile device scans and connects;
When the connection is successful, "Connect!" is output through the serial port, indicating that the Bluetooth is connected.
The serial port outputs prompt information and Pin5 outputs a high level:
Connect!\r\n
```

rusefillc, 2022-01-17:
my test device
https://www.adafruit.com/product/954#technical-details
white RX into USB port, and green TX

rusefillc, 2022-01-17:
No response from WL9981TC even fully removed from Hellen :( No signs of life so far.

[image]

rusefillc, 2022-01-18:
I am starting to wonder if WL9981 is a totally wrong thing for us. They say BLE bluetooth low energy which seems to mean bursts of data communication style not serial port profile SPP at all?

rusefillc, 2022-01-18:
See #23

closed 2022-01-18 by rusefillc

thirstyone, 2022-01-18:
> my test device
> https://www.adafruit.com/product/954#technical-details
> white RX into USB port, and green TX

blocked link (access is denied to russian users probably)

kevindebosscher, 2023-10-01:
Hi,

did you managed to get the WL9981TC working?
Any AT-commands available?

thx

rusefillc, 2023-10-01:
@kevindebosscher nope dropped it sinse no serial profile anyway