rusefi/alphax-2chan #19: FATAL lights up GREEN led?!

rusefillc, 2022-01-16:
I have FATAL in firmware

i have solid GREEN instead of solid RED led
blue is blinking as expected

@andreika-git any ideas?

rusefillc, 2022-01-16:
hm looks good to me

[image]

did I copy-paste totally wrong LED pinout? do we have other hellen boards with wrong pinouts?

@andreika-git next step would be mentioning official LED pinout in meta header for reused by .mk files if that's even possible

andreika-git, 2022-01-16:
> @andreika-git any ideas?

Can you test the LEDs without FATAL, while in normal operation? For example, using the "benchSetPin" Console command on each LED pin...

rusefillc, 2022-01-16:
```
EngineState: pin PG0: led: CRITICAL status

2022-01-16_16_56_47_666: EngineState: pin PG0 set value
2022-01-16_16_56_47_666: EngineState: pin PG0 value 1
2022-01-16_16_56_47_666: EngineState: confirmation_bench_clearpin pg0:18
2022-01-16_16_56_55_405: EngineState: pin PG0 set value
2022-01-16_16_56_55_405: EngineState: pin PG0 value 1
2022-01-16_16_56_55_405: EngineState: confirmation_bench_setpin pg0:16

```
i think FW is changing state of PIN too quickly for me to notice anything?

closed 2022-03-17 by rusefillc