rusefi/alphax-2chan #20: Schematics usability

rusefillc, 2022-01-16:
IN_MAP is IN_MAP2 and IN_MAP2 is IN_MAP3?

that's not amazing schematics readbility in my opinion.

[image]

rusefillc, 2022-01-17:
We cannot change blue names since those are shared Hellen names, let's change black names since those are board specific.

Let's change black "IN_MAP" into "INP_MAP" and black IN_MAP2 into INP_MAP2?

mck1117, 2022-01-17:
What is INP? Why not internal and external MAP?

rusefillc, 2022-01-17:
INP_* is the smallest change to remove all conflicts between black and blue

I agree that INT_MAP and EXT_MAP would be even better black net names!

andreika-git, 2022-02-16:
done
[image]

rusefillc, 2022-02-16:
Thank you that's progress for sure!

Open question in black should better be IN_MAP2 or INP_MAP2 at this point I've lost track of what are the arguments supporting which side, at least we've address IN_MAP2 drama

closed 2022-02-16 by rusefillc