rusefi/alphax-2chan #22: No wake up with TJA1442AT

rusefillc, 2022-01-17:
I've soldered TJA1442AT on one unit with stm32f407 MCU

I have CAN validated see https://github.com/rusefi/alphax-2chan/issues/16#issuecomment-1014172777

I issue "stm32_standby" via rusEFI console

rusefillc, 2022-01-17:
[image]

rusefillc, 2022-01-17:
at the moment we are trying to wake up through op-amp, let's try bypassing manually?

@andreika-git where is a good source of +3v and good poking spot to attempt waking up more manually bypassing op-amps?

rusefillc, 2022-01-17:
Actually code was not ready https://github.com/rusefi/rusefi/commit/54b70fbc3c9cf320f46783310aaea3ea52877c73

mck1117, 2022-01-17:
Should work with opamp

rusefillc, 2022-01-17:
https://github.com/rusefi/rusefi/commit/54b70fbc3c9cf320f46783310aaea3ea52877c73 did not help

@andreika-git please advice on test point for jumper wake up test

rusefillc, 2022-01-17:
Just verified baseline: wake-up via PA0 works on F407 discovery

rusefillc, 2022-01-17:
Hellen does not wake up with +3 or +5 applied to CAN pin 4 with CAN transceiver removed. I am stuck and out of ideas :(

[image]

NMSTEC, 2022-01-17:
Either I'm misunderstanding, or you guys are trying to wakeup an F4 via CAN RX. F4 cannot wake via any pin other than PA0, and rising edge.

rusefillc, 2022-01-17:
This board feeds can_rx into pa0 via op-smp

NMSTEC, 2022-01-17:
My bad, apologies.

rusefillc, 2022-01-17:
great news! discovery firmware on alphax hardware wake up via 5v on CAN RX pad with transceiver removed!

op-amp divider drama - same discovery firmware on alphax hardware does NOT wake up with 3v on CAN RX pad with transceiver removed!

rusefillc, 2022-01-17:
with full erase alphax firmware wakes up on 5v, looks like we had some settings drama or maybe I did not flash latest properly

sounds like next step to address op-amp vs 3v drama. 

i assume divider would not be easy to remove due to 4-resistor-arrays. do we want to replace 4-resistor-array with three individual resistors or is that a bad track overall?

andreika-git, 2022-01-19:
Let's do a little fix:
1) Find a 510 Ohms 0603 resistor on the Hellen72a board:
"510","R212, R213, R214, R215, R116, R119, R120, R124","R0603"
2) Solder it over the 0604-x4 here:
[image]
[image]

rusefillc, 2022-01-20:
I've removed CAN transceiver for first round of experiments

I've soldered resistor on top of array as recommended

first tests stand by was rebooting right away, got that a couple of time

took stuff back for inspection / poking, solder looked good

did more tests and behavior changed, in a good way I guess

it goes to sleep into standby and wakes up with just an antenna - i touch CAN RX pad with jumper wire and that wakes MCU up, even with jumper wire floating

rusefillc, 2022-01-20:
i went forward and re-install can transceiver

now it again reboots after standby right away :(

andreika-git, 2022-01-22:
1) boardPrepareForStop() has palEnableLineEvent() but boardPrepareForStandby() does not
2) PAL_EVENT_MODE_RISING_EDGE --> PAL_EVENT_MODE_FALLING_EDGE ? Rx is "1" by default?

rusefillc, 2022-01-26:
@mck1117 any ideas why does it start right after standby command with CAN transiever?

mck1117, 2022-02-05:
TJA1051

[image]

yellow = CAN_TX
pink = CAN_RX (via opamps, connected to PA0)
blue = CANH

when we shut down the STM32 drags CAN_TX down (ie, dominant), and after ~1ms the transmitter times out (prevents hung bus), when it times out it goes back high and wakes the STM32

rusefillc, 2022-02-08:
@mck1117 I am super confused

do we have same or difference behavior? Do you have desired behavior? Because I believe i have desired behavior?

https://youtu.be/c19ICpoVFuE

rusefillc, 2022-02-09:
forgot to mention that I had a mod to change op-amp divider ratio

[image]

@mck1117 says he had similar mod

rusefillc, 2022-02-09:
just removed the mod and board no longer wakes up, the op-amp divider change is mandatory

andreika-git, 2022-02-09:
We should test another option: use +5V VIO for CAN driver instead of 3.3V - maybe it's better than trying to bypass the opamp divider...

mck1117, 2022-02-09:
That would work (CAN RX is of course 5v tolerant), or we can add a line on the MCU module that goes directly to PA0 instead of thru the opamp.

rusefillc, 2022-02-09:
i believe the open question if we should mod one of alphax boards to triple-check or just fab with 5v CAN VIO without a HW test of rev A

@mck1117 any chance you would be available to test rev A of alpha with +5v VIO?

rusefillc, 2022-02-14:
[image]

rusefillc, 2022-02-16:
Complete failure of my experiment - standby always reboots right away unconditionally

op-amp resistor hack removed
corner lead lifted
red jumper added 5v confirmed in red jumper

same result with following tests:
standby request via CAN
standby request via USB with external CAN device connected
standby request via USB with external CAN device disconnected

[image]

[image]

andreika-git, 2022-02-16:
Does CAN work by itself with the red jumper?

rusefillc, 2022-02-16:
``standby request via CAN`` meaning I have a lua script "onCanRecieve of packet 0x570 sleep' so yes, RX worked

just tested can transmit as well just to be sure and it works

[image]

rusefillc, 2022-02-17:
10K between TX and +5v has changed things but did not solder the overall issue

good news there is no auto-reboot, i.e. "go to sleep" works via CAN now

still broken: no wake up on next CAN message

now I will solder test point for RX

rusefillc, 2022-02-18:
I am failing to do these HW mods that's above my skillset, we need to go a different route.

while adding RX test point I've shorted the board somewhere. I had to remove the resistor and red jumper wire to get board back to life

i got board back to life

i've added red wire and TX pull-up

and now it reboots instead of sleep :(

let me see if @mck1117 can ship his alphax board to @NMSTEC maybe @NMSTEC would be better than me at HW hacks

rusefillc, 2022-02-18:
@NMSTEC says
https://sparkoslabs.com/product/dip-to-soic-op-amp-adapter/
https://sparkoslabs.com/discrete-op-amps/soic-to-dip-adapter/
https://www.sparkfun.com/products/13655

NMSTEC, 2022-02-18:
Just makes prototyping easier with less chance of damaging pads or board. Can also run wires to a breadboard for prototyping. 
Used it often when I needed to constantly pull eeproms off OEM ecus.

rusefillc, 2022-03-30:
See #38