rusefi/alphax-2chan #23: Looks like we have to go HC-0x form factor for bluetooth

rusefillc, 2022-01-18:
still possible to buy good old HC-0x ford factor

JDY-31 is it compatible?
JDY-33 seems to be the dominating one, is it compatible in terms of form factor?

i guess on AlphaX it would go on the back side?

ElDominio, 2022-02-09:
The JD BT boards don't work, I tried both JDY and another BLE compatible HC05 clone, and got it only to work with HC05 (with Realdash)

andreika-git, 2022-02-16:
done
Switched to JDY-33 footprint
[image]

rusefillc, 2022-02-16:
does JDY footprint match HC?

andreika-git, 2022-02-16:
Not entirely, HC also has bottom pads, and JDY doesn't. But we use only several pads on the left/right, and they match. I guess HC will fit and work too.

rusefillc, 2022-02-16:
grrrrrrrrrrrrrrrrrrrrr

closed 2022-02-16 by rusefillc