rusefi/alphax-2chan #25: Remove diodes from injector channels

mck1117, 2022-01-28:
D23 and D24 must be removed. For injectors we want active clamping. We only want freewheel diodes for non-injector solenoids (idle, etc)

rusefillc, 2022-01-28:
@andreika-git do we need to DNP those?

@ElDominio can you please remove from the board you have?

[image]

mck1117, 2022-01-28:
D23/24 are already marked DNP on schematic, looks like it didn't happen. Those footprints can be removed entirely, no reason to keep them.

rusefillc, 2022-01-29:
@andreika-git is this related to https://github.com/andreika-git/hellen-one/issues/78 ?

andreika-git, 2022-01-29:
I don't understand why JLC populated them. Here's the final BOM file with the DNP instead of PN, which was sent to the fab:
[image]

rusefillc, 2022-01-29:
[image]

ElDominio, 2022-02-09:
Will remove and report back, the car turned on with them on.

andreika-git, 2022-02-16:
done, the diodes completely removed
[image]

closed 2022-02-16 by rusefillc