rusefi/alphax-2chan #32: C4 digital input to support more modes

ElDominio, 2022-02-16:
Change hardware setup to make C4 a full digital input, able to receive 12v/5v or ground when pullup is enabled.

mck1117, 2022-02-16:
Hardware should already be a full digital input, no? Permanently enabled pullup, tolerant of +-24V or so.

ElDominio, 2022-02-16:
Some people wire, for example, the positive switch of a nitrous solenoid as a "nitrous engaged" input, so being able to disable the internal pullup is important for usecases like this

rusefillc, 2022-02-18:
rev A status: C4 response to GND (at the moment we always have pull-up)

ElDominio, 2022-02-18:
I'd say C7 is fine being as it is (although the pullup should be selectable, some Chryslers have sensors with built-in pullups), C4 does need more modes

andreika-git, 2022-02-18:
Warning! LEDs will be inversed if pulldown mode is selected for CAM or 2STEP.

closed 2022-02-18 by rusefillc