rusefi/alphax-2chan #34: Pin C1 not working "Fan relay"

ElDominio, 2022-02-17:
Activating pin C1 in rusEFI console does nothing, neither does it when it meets conditions in Tunerstudio. 

The relay for the "fan" does not activate, we disconnect from C1 and put it directly to ground and the relay DOES activate.

rusefillc, 2022-02-17:
please provide a link to you tune

please confirm that you have power cycled ECU

please clarify how do you test via console

update: please provide more details on how exactly does it crash console maybe even separate tickets since looks to be a console issue

rusefillc, 2022-02-17:
@ElDominio those four mosters are soldered by me, i.e. huge risk of poor soldering

can you try another board?

ElDominio, 2022-02-17:
1) latest tune is on rusEFI online

2) i did

3)  so it was a firmware mismatch, was there a firmware upload recently? i updated everything and it connects
So I just go to Bench Test on RusEFI online and hit "Test" on Radiator Fan, no bulbs light up on the rusEFI

rusefillc, 2022-02-17:
:( "please go find my latest tune hopefully you would find the right one"

rusefillc, 2022-02-17:
3. I do not follow. "connects". is this about console issue or C1 issue? it's very ineffective to discuss two separate issues here

with latest TS your TS asks even offers you to download proper .ini Are you using latest TS .07 or .08?
firmware changes 6 times a day
console does not automatically sync with firmware :(

ElDominio, 2022-02-17:
https://rusefi.com/online/view.php?msq=877

When i say "it connects" means rusEFI console connects to the 2chan. That was resolved by firmware stuff.
There is no console issue. It's C1 that's not working right now