rusefi/alphax-2chan #39: Document all pullup/pulldown lines

rusefillc, 2022-03-13:
We need to document what state of pullup/pulldown GPIO lines does what

[image]

for instance, CRANK_*_PULLUP set to 0 is it VR while CRANK_*_PULLUP set to 1 is Hall?

I believe we need explicit readable text maybe table right on schematics PDF

andreika-git, 2022-03-13:
We have two separate "pieces of knowledge" here:
1) CRANK_PULLUP set to 1 means that pull-up resistor is enabled, while set to 0 means the pull-up is disabled.
2) Hall mode needs pull-up resistor enabled while VR mode doesn't.

So technically speaking, both VR and hall are always "enabled". And CRANK_PULLUP doesn't switch between them, it just helps them to work correctly.

rusefillc, 2022-03-13:
@andreika-git at the moment we have no user facing summary of what are the five options doing.

@andreika-git at the moment I do not know which value enables what user experience myself.

it looks like for crank 0=VR and 1=Hall from user perspective

need same 0=X and 1=Y explanation for all other options

like thermistors,
0="Use build-in pull-up" and 1="No pull-up" or the other way around?

mck1117, 2022-03-13:
>     2. Hall mode needs pull-up resistor enabled while VR mode doesn't.

And part 2b is that for some hall crank sensors, it doesn't actually matter whether you have a pullup or not. My LS cam/crank sensors output a nice sharp 12v square wave, no pullup required.

closed 2022-03-17 by rusefillc