rusefi/alphax-2chan #40: C4&C7 pull down mode does not work on rev B

rusefillc, 2022-03-14:
C4 and C7 pull-down test FAIL (?)

regardless of pull-down C7 triggerPrimaryRise increments when I touch +5v and does not increment when I touch GND

measures 0.9v with pull-down=true
measures 1.3v with pull-down=false

rusefillc, 2022-03-14:
pull-down mode:

R29 side towards main connector 0v
R29 side towards USB  connector 4.8v

R43 side towards main connector 3.2v
R43 side towards USB  connector 0v

pull-down up:

R29 side towards main connector 0.6v
R29 side towards USB  connector 4.8v

R43 side towards main connector 0.0v
R43 side towards USB  connector 0v

rusefillc, 2022-03-14:
@andreika-git suspects HW issue around Q20 & Q21

andreika-git, 2022-03-15:
[image]

rusefillc, 2022-03-15:
[image]

rusefillc, 2022-03-15:
LED works as we expect it to work - in inverted mode with one of operation modes, but main thing LED goes from ON to OFF or from OFF to ON

closed 2022-03-17 by rusefillc