rusefi/alphax-2chan #41: Pull-ups 5VA or 5VAS in case of TLS115

rusefillc, 2022-03-17:
If we go without R40 with TLS115 do we need all pull-ups on the samve 5V* or not?

[image]

mck1117, 2022-03-17:
what is R40

rusefillc, 2022-03-17:
@mck1117 

[image]

rusefillc, 2022-03-26:
this breaks into two questions

1) safety in terms of being exposed
2) power management i.e. which 5v is turned off when board goes stand by

rusefillc, 2022-08-30:
@andreika-git poke

andreika-git, 2022-08-31:
TODO: add explanation for +5VAS on the schematics.

andreika-git, 2022-09-01:
[image]

closed 2022-09-01 by andreika-git