rusefi/alphax-2chan #46: Silkscreen IGND reads as "I-GND" not as "IGN-D"

rusefillc, 2022-04-01:
See https://github.com/rusefi/alphax-4chan/issues/15

andreika-git, 2022-08-30:
@rusefillc Can you please clarify this issue? We do not have "IGND" on 2chan.

rusefillc, 2022-08-30:
i think it's IGNB on silk which confuses me

so pinout says B

[image]

while schematics says 1-2-3-4?

[image]

@ElDominio do you want anything touched? which way do you want to touch things? shall we mention "B" on schematics? shall we change silk from IGNB to something with "2"?

ElDominio, 2022-12-03:
Close?

closed 2022-12-03 by rusefillc