rusefi/alphax-2chan #50: Pin C4 not behaving correctly? Need testing?

ElDominio, 2022-04-19:
I have both a revA and a revB with issues on the Launch Control input where it stays engaged no matter what. I haven't been able to test much, but Tercel and Corolla both stay with launch control engaged if I enable launch control. Will update with more tests, but if any bench tests can be done it would be great

andreika-git, 2022-04-19:
What's the value of the setting 'boardUse2stepPullDown'? Is it "With Pull Down" or "With Pull Up"?

ElDominio, 2022-04-19:
1) Pretty sure I have it as "With Pullup", since we wired it to a switch to ground

2) Have not tested revC yet

andreika-git, 2022-04-19:
1. Does the revB board have the required fix of Q21 transistor? (see https://github.com/rusefi/alphax-2chan/issues/40)

2. Did you set the Launch Control settings correctly? 
  a) "Activation Mode" should be set to "Switch Input"
  b) "Switch Input" pin should be set to "C4 - 2STEP"

Currently there is no option for "Switch Input inverted" in the firmware, although we already have "Clutch Input inverted". That's why your switch may act in inverted mode, and launch control may be engaged all the time except when activated. Maybe we should ask @rusefillc to add the inverting pin mode option...

3. Can you please try the "readpin PF11" command in the rusEFI Console? Please run it twice with the switch enabled and disab led. This will check the voltage on the input pin without any Launch Control logic.

rusefillc, 2022-04-19:
@ElDominio I've made the change we can now adjust Launch control pin mode

Please update to latest firmware, please select inverted pin mode, please report back.

rusefillc, 2022-04-19:
@andreika-git reminded me that we need actual code for that not just UI :)

ElDominio, 2022-04-19:
I updated, used it as clutch input and now it works fine
[image]

andreika-git, 2022-04-19:
It would be great if you could test with the new "Switch input inverted" option too...

ElDominio, 2022-04-19:
[image]

it disables as an option if i use switch input mode

andreika-git, 2022-04-19:
We'll fix it, thanks!

ElDominio, 2022-04-19:
[image]

That is fixed, but in switch mode it still doesn't listen to what I select. It is active when 5v is present, inactive when GND, and I need the opposite

ElDominio, 2022-04-19:
[image]

When I set it as clutch input, it sets the one as clutch down too and it throws a critical error lol

ElDominio, 2022-04-19:
[image]

If I set it as clutch input, the settings in Misc sensors gets auto-set, and rusEFI throws a critical error

[image]

rusefillc, 2022-04-19:
its greyed but still selected. you need to fully unselect when changing mode

rusefillc, 2022-04-19:
OK so original problem is that we launch button "inverted" is missing! 

should be
[image]

and we have

[image]

rusefillc, 2022-04-20:
@ElDominio just added logic to hand the new "inverted" flag.

Overall we have three things for each switch:
1. pin
2. inverted or not
3. physical pull ups

rusefillc, 2022-04-22:
@ElDominio how does it look with latest changes?

rusefillc, 2022-04-27:
closing for now?

closed 2022-04-27 by rusefillc

ElDominio, 2022-04-27:
New update is great and works great 😁