rusefi/alphax-2chan #52: smoke on rev A

rusefillc, 2022-04-29:
[image]

[image]

rusefillc, 2022-04-29:
@ElDominio any additional info?

rusefillc, 2022-04-29:
all these pull-ups are NOT protected from +5v on the wire while ECU is not powered by +12v

Overall need more info :)

ElDominio, 2022-04-29:
Was used to run Tacho in 12v mode.

Was on Corolla 1ZZ 2003

The ECU also doesn't read RPM anymore, maybe thanks to this?

rusefillc, 2022-04-29:
Yes this is about tach for sure

Do you have a wiring diagram for that 2003 1ZZ?

ElDominio, 2022-04-30:
All the info I have at this point:

Car was running fine in the shop, tach was also working fine, set as 12V. 
Owner has some odd cranking issues the next day, tech tells owner to check a few wires
Car never started again
Car taken to our shop, tech finds damage on ECU.

The car has very bad sync issues now (VR crank). Could anything here damage anything sync related?

I will be at the shop on Tuesday

mck1117, 2022-04-30:
@andreika-git what change was made in your commit? Short description in commit message would be great...

It's possible we exceeded the maximum allowed V_gs on the mosfet - the maximum allowed is +-12v, which we exceed if the pullup is enabled and the alternator charges the battery at 14-15v or whatever. It's certainly at least marginal.

ElDominio, 2022-05-01:
The strange this is it ran fine until the owner took it home... is this something that would take a while to kill?

rusefillc, 2022-05-01:
if it's about MOSFET dying from voltage
@mck1117 says

The mosfet may have
1 million hours failure at 12 volts
1000 hours at 13v
10 hours at 14v
1 minute at 15v
1 second 16v
0.001 second 20v

andreika-git, 2022-05-01:
> @andreika-git what change was made in your commit? Short description in commit message would be great...

The idea of my fix is: if +12V_RAW is not a power source but a load (when connected to a car harness), then high current may flow through Q3 mosfet from 5V to 12V...
[image]

mck1117, 2022-05-01:
@andreika-git yes, that's also a possible answer.

This would solve both issues:

[image]

1. add a diode between Q3 and D28_1, so that current can only flow out of the pullup.
2. Add a zener between Q3 pins 1 and 3 to limit V_gs
3. Add a resistor between Q1 and Q3 to limit current in the zener in (2)

ElDominio, 2022-05-01:
Assuming this happened, could it take down the processor with it? Car has very bad sync issues (sometimes reads, but it never reads enough to run)

ElDominio, 2022-05-02:
Engine not running was due to incorrect gap override, @mck1117  advised and that was fixed, the car runs again.

Question, would using tach output at 5v and using that signal for an external circuit work? could this ECU still output a 5v wave?

mck1117, 2022-05-02:
I thiiiiiink that would work? If Q3 failed open-circuit, then it should still work as a 5v pullup output.

ElDominio, 2022-05-02:
Checked and that's quite probably correct. The output still outputs a 5v wave, so we'll use that with an external 2222 to drive the tacho, thanks

rusefillc, 2022-05-02:
@andreika-git do you want to upgrade Q3 similar to part suggested at #53

mck1117, 2022-05-02:
> @andreika-git do you want to upgrade Q3 similar to part suggested at #53

my recommendation above about adding a zener would mean the current Q3 is OK, and would protect against transients above 20v anyway

ElDominio, 2022-05-03:
BTW

We did the external circuit with a 2222 and it's working fine again :)

andreika-git, 2022-05-03:
I think @mck1117's method is more universal and preferred.

andreika-git, 2022-05-03:
On the other side, we also have other p-channel mosfets:
[image]
And it seems we don't have free space for fitting additional components for these (without major pcb redesign)...
Maybe we could limit the +12V source used for pull-ups? We could use a free PSM712 (TVS), which would allow to use SI2347DS mosfets. Of course +$3 will be added to the price...

andreika-git, 2022-05-03:
Something like this:
[image]

mck1117, 2022-05-05:
I see you've also added a resistor divider to the gate of Q3 - that's probably enough to avoid problems. The TVS probably isn't necessary, as that circuit is now 100% good up to at least 24v input.

andreika-git, 2022-05-05:
> I see you've also added a resistor divider to the gate of Q3 - that's probably enough to avoid problems. The TVS probably isn't necessary, as that circuit is now 100% good up to at least 24v input.

Yes, but the TVS is to protect other p-channel mosfets: q4, q5, q16, q17 - they didn't have a spare resistor, and it's not easy to find a free space for additional ones:
[image]
But maybe you're right and I should try harder and solve the problem by squeezing in the resistors...

andreika-git, 2022-05-09:
Added more gate resistors. Now we can use our favorite AO3401 as before.
[image]

rusefillc, 2022-12-01:
i guess time to close?

closed 2022-12-01 by rusefillc