rusefi/alphax-2chan #62: ONE revC does not read TPS voltage

ElDominio, 2022-08-23:
Raw TPS gauge shows 0v, swapping to a known working unit (revB it reads fine (1.7v) on my test bench. What can I check?

I think all other revC's have worked, i'll have to see if I have anything on hand to swap with

rusefillc, 2022-08-23:
@ElDominio if we care about this unit my suggestion is to ship it back to me. Overall it would be quicker for me to mess with it in NJ

ElDominio, 2022-08-23:
I do have capable people of soldering and I can take measurements, if there's anything that could be suspicious it would make it faster for me to find.

rusefillc, 2022-08-23:
Anything could be equally broken, it's just the boring tracing from where you have voltage towards where you do not based on schematics.

ElDominio, 2022-12-03:
Has been shipped to @rusefillc

rusefillc, 2022-12-29:
it's JLC fab defect, this board now works with a bit of solder on op-amp

closed 2022-12-29 by rusefillc