rusefi/alphax-2chan #66: U1 TLS115

rusefillc, 2022-10-04:
[image]

Did we forget U1 BOM? :(

rusefillc, 2022-10-04:
#64

ElDominio, 2022-10-05:
https://github.com/rusefi/alphax-4chan/blob/main/bom_replace_alphax_4ch-d.csv

Not in BOM, and I cant figure which one it is

[image]
 Not in stock in Digikey if this is it

rusefillc, 2022-10-05:
we use 10 pad package here not the 8 pad package

digikey TLS115D0LDXUMA1 please

rusefillc, 2022-10-05:
@andreika-git please fix master kicad files and fix https://github.com/rusefi/alphax-2chan/blob/main/boards/alphax_2ch-d/board/alphax_2ch-d-BOM.csv manually

closed 2023-01-28 by rusefillc