rusefi/alphax-2chan #67: conditioned crank test pad prior to rev D

rusefillc, 2022-10-08:
@ElDominio needs a way to hack a crank test pad prior to rev D https://github.com/rusefi/alphax-2chan/issues/63

rusefillc, 2022-10-08:
so we need some resistor or capacitor either on the output of vr-descrete or on IN_CRANK input module input side

[image]

rusefillc, 2022-10-08:
[image]

rusefillc, 2022-10-08:
@ElDominio that bright green LED looks like it's the output, it's apparantly supposed to be blinking with crank?

one of the sides of it or maybe any side would make a great test point

ElDominio, 2022-10-08:
i love you

closed 2022-10-08 by rusefillc