rusefi/alphax-2chan #76: rev E test log

rusefillc, 2023-01-05:
* PROG button works - blue LED blinks
* works from external +12v power supply seems happy! https://github.com/andreika-git/hellen-one/issues/246
* +5 goes out
* TPS sensor works - OPEN QUESTION WHAT analog mux considerations we have here?!
* two injectors are clicking in bench test
* SD card does NOT work so far probably firmware needed?

rusefillc, 2023-01-05:
Fun fact - rev A has on-board baro (another one, not the on-board MAP) and on-board accelerometer (maybe without firmware support?!)

Qwerty-OFF, 2023-01-05:
SD card is now connected to SPI1

rusefillc, 2023-01-22:
@ElDominio did you have a chance to put one on a real vehicle? VR vehicle or non VR vehicle?

ElDominio, 2023-01-22:
Tomorrow, dual hall

rusefillc, 2023-01-22:
Need a VR test as well :)

mck1117, 2023-01-22:
> Need a VR test as well :)

All VR since forever is broken without the extra caps bodged on, which should happen to every pre-0.4-vr ECU out there if you want to use VR on it.

rusefillc, 2023-01-22:
@mck1117 somehow we had happy VR experience on some cars?

@ElDominio did we have _any_ happy VR vehicles on Alpha?

mck1117, 2023-01-22:
One edge was better than the other. If you are wired to use the good edge and lucky enough, it'd usually work without the caps. Unlucky or wired backwards you need the caps to get a usable signal at all.

rusefillc, 2023-02-13:
@ElDominio poke need real vehicle 2chan rev E results please

ElDominio, 2023-02-13:
Once I have them on my hands

closed 2023-08-08 by rusefillc