rusefi/alphax-2chan #85: Digital input issue status clarification on rev E with mm-0.1

rusefillc, 2023-02-22:
@mck1117 is looking to clarify status of digital input on mm, let's use this 2chan rev E as an example

[image]

Context https://github.com/rusefi/alphax-8chan/issues/28#issuecomment-1290187356

Context https://github.com/rusefi/alphax-4chan/issues/72

```
help me understand how megamodule helps this input issue
I see no digital input buffers on megamodule
```

mck1117, 2023-02-22:
Ok but that's not inside the megamodule?

rusefillc, 2023-02-22:
Ohh I see. We hope to have it on the _boards_ but question if it should be part of _mm_.

rusefillc, 2023-02-22:
@mck1117 
```
I don't know why it isn't its own module

```

Qwerty-OFF, 2023-02-22:
We have 4 Schmitt trigger cannels in MM that called IN_D1...IN_D4:
[image]
In 2chan for VR compatibility R140 = 0 Ohm, used external large resistors:
[image]