rusefi/alphax-2chan #86: revF wishlist

ElDominio, 2023-03-11:
My requests for "final" revision of 2chan

-> USB-C port instead of mini USB

-> 8 pin Molex microfit internal with additional I/O (internally on PCB)
[image]

things on the molex:
- 1x injector
- 1x ignition
- 2x analog in
- 2x digital in
- 2x grounded output

-> Built in baro onboard, so this menu is used
[image]

This one? https://jlcpcb.com/partdetail/Goertek-SPL06001/C2684428

ElDominio, 2023-03-15:
Move MAP location nearer to the middle of the board or close to connector, to evade damage during shipping

[image]

Like so
[image]

rusefillc, 2023-05-08:
@Qwerty-OFF does 2chan need any VR/hall magic on top of rev-E? or is it that all hard cases are 4chan anyway?

ElDominio, 2023-05-08:
I think revE is working quite well, have not ween a need for anything more than we have right now?

rusefillc, 2023-06-11:
Ordering rev F now

closed 2023-06-11 by rusefillc