rusefi/alphax-2chan #96: CRITICAL: 2chan revE started, not pulsing Coil1 or 2 after running for a few seconds

ElDominio, 2023-05-14:
https://github.com/rusefi/alphax-2chan/assets/5685108/716a2cc6-083f-4eb1-a867-26d8ce000559

This is all I have since they are beginning engineering students

I made this simulation log, I see coilstates changing, but physically only can see 2.5 when pulsing 2seconds on/off in Bench Test mode (no 5v)

[simulationlog.zip](https://github.com/rusefi/alphax-2chan/files/11470845/simulationlog.zip)

ElDominio, 2023-05-14:
update;

channel 1 is back to life, but channel 2 is still dead. What can they check?

rusefillc, 2023-05-14:
do they have a digital multimeter with frequency measurement? they should play with self-stimulation and DMM reporting frequency at proper relevant points

ElDominio, 2023-05-14:
Self stim at what RPM? I think they can do that

rusefillc, 2023-05-14:
600
1200
2400

Frequency should change accordingly

Qwerty-OFF, 2023-05-14:
Is only one board tested? We have duplicate LEDs in mega-mcu144, do they behave the same way?

[image]

ElDominio, 2023-05-14:
This board is with a Formula SAE team in Mexico so not easy to test heh

They reported the engine ran for about 10 seconds, then ignition channels died. 

They are using one of these igniters: https://www.efihardware.com/products/2270/Bosch-4-Channel-Ignitor

Is there any component that can be replaced or any point that can be probed? I will take one of my igniters to the FSAE competition (and a spare ECU) but it's a bit odd.

Qwerty-OFF, 2023-05-14:
You can check the signal next to U460. On the right are inputs from the MCU, on the left are outputs through series resistors of 10 ohms. If there is a signal on the right and none on the left, replacing U460 might help.

Does the team use one output to simultaneously control two coils?

[image]

Qwerty-OFF, 2023-05-14:
You can also check the self-recovering fuses F1 and F2.
[image]

ElDominio, 2023-05-14:
> You can check the signal next to U460. On the right are inputs from the MCU, on the left are outputs through series resistors of 10 ohms. If there is a signal on the right and none on the left, replacing U460 might help.
> 
> Does the team use one output to simultaneously control two coils?
> 
> [image]
> 
> 

Yep since it's a 4 cylinder

ElDominio, 2023-05-19:
[image]

I believe this is somehow the culprit, have the ecu on hand for testing

ElDominio, 2024-02-23:
It was the culprit, and the coi2 output died along with it

closed 2024-02-23 by ElDominio