rusefi/alphax-2chan #97: ibom 3D PCB view degradation

rusefillc, 2023-06-11:
https://rusefi.com/docs/ibom/alphax_2ch-e-ibom.html click PCB - stm32 is displayed properly
https://rusefi.com/docs/ibom/alphax_2ch-f-ibom.html click PCB - we have stuff shifted, we now have main connector which we did not see before?

rusefillc, 2023-06-11:
@Qwerty-OFF ты знаешь, где трогать - или надо @andreika-git трогать?

Qwerty-OFF, 2023-06-11:
Я в кикаде переставил в левый нижний угол метки начала координат, которые почему-то были не в нем. Допускаю что дело в этом, но что делать с этим не знаю
[image]

rusefillc, 2023-06-11:
@andreika-git нужна может быть только документация, без помощи?

andreika-git, 2023-06-11:
После переставления начала координат нужно при экспорте VRML убедиться, что указанные там координаты совпадают с этим началом

andreika-git, 2023-06-11:
https://github.com/andreika-git/hellen-one/wiki/fix-image-offset-howto

andreika-git, 2023-07-07:
Fixed!
There were several issues: wrong image offset and wrong offset of the main connector's 3d-model.

rusefillc, 2023-07-07:
https://github.com/rusefi/alphax-2chan/commit/12e65748e2b58275bfda090a919dd1a521cffd58 touches only total offset unless I am missing something? where do we touch main connector 3D model?

andreika-git, 2023-07-07:
> [12e6574](https://github.com/rusefi/alphax-2chan/commit/12e65748e2b58275bfda090a919dd1a521cffd58) touches only total offset unless I am missing something? where do we touch main connector 3D model?

It's inside the .kicad_pcb file. The 3d model itself stays the same, it's the offset used for the component was wrong:

[image]

closed 2023-07-07 by rusefillc