rusefi/alphax-2chan #108: CRITICAL: ECU does not power on correctly with 12v, inserting USB makes ECU happy  until next key on

ElDominio, 2024-04-05:
Video of problem here: https://youtube.com/shorts/LHZ4HpVnu9k

Seems the ECU is not waking up correctly on 12v key, once USB is connected it wakes up and stays running fine until next power cycle

rusefillc, 2024-04-05:
@ElDominio any context?

all 2chan boards on all vehicles?

specific board on specific vehicle?

has it worked before?

how do I reproduce this in NJ?

can you reproduce this with wires without a vehicle?

ElDominio, 2024-04-05:
RevG

Car was on revC and everything was fine

i have no answer to the other two questions, do you not have a revG board on hand?

BTW i tried full chip erase and the problem persists

ElDominio, 2024-04-05:
I had him power on the unit externally and it powered up correctly; they fucked up wiring. Forgive me for using critical tag i was in panic

closed 2024-04-05 by ElDominio

ElDominio, 2024-04-05:
just to update: it was using an external VR converter of my design (uses 5V, gnd, vr+/- and goes to CKP input on ECU); somehow removing it made it work fine. Are there any theories why having it hooked up made the ECU mad?