rusefi/alphax-2chan #112: RevG with boot loop

ElDominio, 2024-09-25:
I have a revGboot looping if powered from 12V, but works fine if first powered by 5v. 

I have tride Full DFU Erase and reloading firmware, but not improvement

rusefillc, 2024-09-26:
@ElDominio I see a statement. I approve this statement.

What is the question? Are you looking to check if all 2chan rev G do the same? Something else?

ElDominio, 2024-09-26:
Yes, I would like if others do the same, and if this is a software or hardware problem

ElDominio, 2024-09-29:
I found an old revG with 2023 firmware, and it was booting up fine. Updated to latest and revG does not boot loops and RARELY finishes booting. RevF boots fine

rusefillc, 2024-09-29:
@ElDominio I have rev G and I am able to reproduce the issue. I have to confess that I was totally NOT expecting this to be a firmware issue.

@ElDominio I need help. Can you please run a binary search between official releases and/or daily builds and attempt to isolate the issue?

ElDominio, 2024-09-29:
I've already downgraded to Dec2023 and still boot loop :(

ElDominio, 2024-09-29:
Feb2023 is ok!

ElDominio, 2024-09-29:
June2023 is broken

rusefillc, 2024-10-06:
2023-03-09 good

https://github.com/rusefi/rusefi/releases/tag/2023-04-03 OK

https://github.com/rusefi/rusefi/releases/tag/2023-04-10 OK

https://github.com/rusefi/rusefi/releases/tag/2023-04-16 OK

https://github.com/rusefi/rusefi/releases/tag/2023-04-17 OK

https://github.com/rusefi/rusefi/releases/tag/2023-04-18 bad

2023-04-19 bad

rusefillc, 2024-10-06:
https://github.com/rusefi/rusefi/tree/2chan-revG

https://github.com/rusefi/rusefi/tree/branch-2023-04-20-stamp

rusefillc, 2024-10-06:
17 in file name is 16 in date_stamp?

firmwareInfo="rusEFI+v20230416%4054879" nPages="1" signature="rusEFI master.2023.04.17.alphax-2chan.3461928810"

https://github.com/rusefi/rusefi/releases/tag/2023-04-17 
[image]

rusefillc, 2024-10-06:
smoking gun https://github.com/rusefi/rusefi/commit/b135fecfe9f61562ff655c86f236f5079e949457

rusefillc, 2024-10-06:
[image]

rusefillc, 2024-10-06:
rev F board is happy with latest firmware and 12v power supply

```
2024-10-06_11_23_51_236: EngineState: hellenBoardId=115
2024-10-06_11_24_11_341: EngineState: pin PE10 ((null)): EN Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
```

rusefillc, 2024-10-06:
rev G is NOT happy with latest fw if started from 12v
```
2024-10-06_11_24_50_336: EngineState: hellenBoardId=210
2024-10-06_11_25_24_413: EngineState: pin PE10 ((null)): EN Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
```

rusefillc, 2024-10-06:
@Qwerty-OFF https://github.com/rusefi/alphax-2chan/issues/100

rusefillc, 2024-10-06:
@ElDominio shit that's https://github.com/andreika-git/hellen-one/issues/353

please remove two capacitors from rev G boards to have them boot. So far our software attempts to fix have failed

rev F i have it happy
[image]

rev G is happy after removal
[image]

ElDominio, 2024-10-06:
fix confirmed!!!b

ElDominio, 2025-03-03:
This is fixed, closing

closed 2025-03-03 by ElDominio